nguy/artview #133: Testing

nguy, 2016-02-29:
Add testing capabilities for future releases.

gamaanderson, 2016-03-02:
This have crossed my mind some times, but I don't see how this can be automatised since unlike pyart this is a grafic tool. We do have an API that could help on that, but we would need to start paying more attetion to it then we are doing now, and of course attention is a limited resource.
@jjhelmus you do spend a lot time creating the automatic tests, don't you?

jjhelmus, 2016-03-02:
Writing and maintaining tests does take time initial but I think in the long run it saves time by reducing the number of bugs and time spent tracking these down.

nguy, 2016-03-02:
Well I'll admit this is a weak spot for me. I have not formulated exactly how to test. But it occurred to me again when looking through Py-ART recently. I also want to make sure we have a relatively stable core before implementing testing.