nguy/artview #134: File Naviagator broke in version 1.2.3

pfhein, 2016-03-01:
`File navigator` is broken in version 1.2.3.  When I click on `File navigator`, this error occurs and no window is opened.
```
starting component: FileNavigator
Traceback (most recent call last):
  File "/usr/local64/python-2.7/anaconda/lib/python2.7/site-packages/artview/scripts/_common.py", line 79, in <lambda>
    lambda mode=mode: MainMenu.change_mode(mode['action']))
  File "/usr/local64/python-2.7/anaconda/lib/python2.7/site-packages/artview/components/menu.py", line 359, in change_mode
    components[i] = components[i](name=name, parent=self)
  File "/usr/local64/python-2.7/anaconda/lib/python2.7/site-packages/artview/components/navigator.py", line 89, in __init__
    self.setWindowState(Qt.WindowActive)
NameError: global name 'Qt' is not defined
```
If I click on `File navigator` again, I do not get an error message but there is still no window.  I do get
```
linking FileNavigator.Vradar to Menu.Vradar
linking FileNavigator.Vgrid to Menu.Vgrid
linking FileNavigator.Vfilelist to Menu.Vfilelist
```
Which I suspect is good.

pfhein, 2016-03-01:
PS My ARTview Tour has been updated.
http://radarmet.atmos.colostate.edu/software/artview/

nguy, 2016-03-01:
@pfhein thanks for the heads up. I just pushed the fix. Some accidental deletion in there it would seem.

Could you pull the latest and test. I added a line that may help to bring the the menu to the forefront on some systems. It does not work on my Mac, but would be good to test on other OS.

nguy, 2016-03-01:
Awesome update on the ARTview tour @pfhein

pfhein, 2016-03-01:
`File navigator` is working now.  The Menu Window is still comes up behind the Display Windows under Scientific Linux 6.

nguy, 2016-03-01:
Thanks, I'll keep an eye out for a solution, but I don't have high confidence at the moment.

closed 2016-03-01 by nguy