nguy/artview #135: Discussion: version 1.3

gamaanderson, 2016-03-10:
After a release it is always go to re-think a little about what we want from the new version.

nguy, 2016-03-10:
Here's a list I'm thinking of (not complete or maybe feasible):
[ ] Add text to plots for image creation
[ ] Collection of files - either something like discussed in Py-ART [issue #435](https://github.com/ARM-DOE/pyart/issues/435) or something internal to ARTview. I still think this would be most useful in Py-ART. But we'll need to think about the interface  in ARTview regardless.
[ ] See about fixing the Menu window being buried. I haven't found a way to do this yet.
[ ] Generate underlying map? Probably a one-off step (assuming no motion) and loading the image behind each time.

nguy, 2016-03-10:
Do we need to rethink the documentation? Maybe save it into the gh-pages branch? Is that a good method?

nguy, 2016-03-11:
I completely agree. I want to keep updating the documentation. I see modifying this as a long run and maybe we could set out specific parts for future releases.

gamaanderson, 2016-03-14:
I just update the list with some thing of issue #107 that I know how to do

nguy, 2016-03-18:
Another potential addition is making use of discrete colorbar option [Py-ART #468](https://github.com/ARM-DOE/pyart/pull/468).

gamaanderson, 2016-03-18:
this issue #126

nguy, 2016-03-18:
Sure is, sorry about that. Had a user request for this last week as well. So it is something desirable.

bdolan44, 2016-03-29:
I'd like to request the option to remove a region of data. Here you would draw a region polygon (much like if you are unfolding data) and have the option to delete data inside /outside the region. This is useful for removing second trip or side lobes, etc. Thanks! PS we are having the radar class use art view to unfold some velocities, so there may be a bit of feedback in the next few weeks!

gamaanderson, 2016-03-29:
thank you @bdolan44 and I'm looking forward for your feedback.

Take a look in this [tutorial](https://www.youtube.com/watch?v=VXUZBiA3HfU). You can just remove just from the inside. If desirable we can also find a way to remove from the outside.

bdolan44, 2016-03-29:
That's perfect, thanks for the tutorial @gamaanderson! 

I have a couple of other comments for now. I'm not sure if this it the right place to put them, but since I'm here, I'll just list them out.
-On my Mac when I open artview it just gives me a big black window to select the files (see attached image). I can actually double-click to open files sort of randomly and finally get into the Display 1 and Display 2, but if I try to write out a file, the file selector box is also black. I just updated this morning (with conda update -c jjhelmus artview). [By the way, under the instructions on the main page for updating, it says 'ARTview' but it should be 'artview' if you followed the instructions for the conda install package].
-Secondly, we think it would be nice if you could leave the region polygon in place between sweeps for editing. For example, my second trip was in the lowest 2 sweeps, but if you just go up in elevation angle, the polygon remains drawn but if I hit "Filter Gates in the Radar" on the 2nd elevation angle it doesn't do anything. We found a work around which is if you 'remove vertex' then close the polygon again, then you can use the same polygon again at the next elevation.
-Third, if the variable is not recognized (I have a 'Vel' field) and you set everything up in Display 1 to have a velocity colorbar and range, then go to Modes -> Manual Unfold, it resets everything and you have to set it up again. This is pretty annoying.
-Fourth, I would be strongly in favor of adding a discrete colorbar option as is being discussed. I just want to add my support for that feature. :)

Otherwise this is great and it's really useful! Thanks!

[image]

nguy, 2016-03-29:
@bdolan44 Did the file window show previously or has this happened only since you updated through the conda channel?

bdolan44, 2016-03-29:
Well....artview used to work fine for me (like earlier this week) but I've been having some other issues with pyart so I updated a number of things this morning, including numpy, pyart, and artview. So I can't say for sure if it was just due to updating artview with the conda channel. :-/

gamaanderson, 2016-03-30:
- artview doesn't do in the file selection it self, this is pure Qt. Try the following script, if it present the same behaviur this is probably a conflict between Qt and your operation system. May be you have some color theme set up etc.

        from PyQt4 import  QtGui, QtCore
        app = QtGui.QApplication()
        QtGui.QFileDialog.getOpenFileName()
        app.exec_()
        exit()
    there are some option that we can pass Qt for solving this kind of thing if we can figure out what is the problem. 

- the select region is still there, but the filter is out dated. What you are doing is forcing it to recreate the filter. I will see what we can do about that.

- I was award of this and it is a side effect of the implementation. However you gave me the idea of adding a lock in the colormap (so components we be asked no to change the colormap, unless they have a reason for doing so). Thank you for the idea.

- I will make that a priority.

bdolan44, 2016-03-30:
Ok...I'm not entirely sure what is going on because I tried to run the script but I think it failed? 

>>> from PyQt4 import  QtGui, QtCore
>>> app = QtGui.QApplication()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: arguments did not match any overloaded call:
  QApplication(list-of-str): not enough arguments
  QApplication(list-of-str, bool): not enough arguments
  QApplication(list-of-str, QApplication.Type): not enough arguments
>>> QtGui.QFileDialog.getOpenFileName()
QWidget: Must construct a QApplication before a QPaintDevice
Abort
app.exec_()
exit()

But I restarted my machine this morning and now I don't have the black window for the file selector with artview. So, problem solved? :-/

gamaanderson, 2016-03-30:
Well, if it is solved, so ok. The script error is my fault, it should be `app = QtGui.QApplication([])`.

bdolan44, 2016-03-30:
Thanks for your help @gamaanderson! Must have been something specific to my machine. But I'm up and running again with artview.

nguy, 2016-03-30:
Good news! We'll look into these other things you brought up as well @bdolan44 . Would you be willing to create individual issues on the Github page when other issues come up in class? It's pretty exciting for us to hear that ARTview is being used for looking at data for a class.

bdolan44, 2016-04-01:
@nguy Will do!

gamaanderson, 2016-05-05:
@bdolan44 PR #142 added a lock in colormap to avoid redefinition and also a new plugin for discret colormaps

gamaanderson, 2017-01-29:
We are mostly done with code part for version 1.3, there is just a discussion to do about the layout. There is however a lot a work to be done in explaining all that is new. So it may take a while to launch, but it is coming.

gamaanderson, 2017-01-29:
by the way @bdolan44 you comment about the select region kind of felt out of the radar (hihi), but I just addressed it now. It was actually pretty easy, I love when good programming decision in the past pay out in the present.

nguy, 2017-06-29:
Is the top list still up to date? My thinking is that we should kick the Collection item to v1.4 down the road and work on getting the package released for August. 

So as I see it we need to
[ ] Update Documentation
[ ] Final code check
[ ] Update README if necessary
[ ] Create a conda release

Anything else?

gamaanderson, 2017-06-29:
as said I think we are out of time for the documentation, and I don't see it helping to many people in the way it is now. So by update documentation I pretty much only mean runing the make.

nguy, 2017-08-29:
This will be closed as PR #202 was merged. A new Issue tracker for v1.4 will be opened.