nguy/artview #137: Add Discrete Color Bars

stechma2, 2016-03-21:
It would be useful if discrete color bars could be used (and easily created, if possible). For example, I wanted a discrete color table to visually identify possible ground contamination in tail radar sweeps. Obviously discrete color bars are desired in many other situations as well.

I used the following code to add a quick and dirty discrete color bar to the ARTview displays, though as implemented, it hard codes the color table, preventing it from being changed within the GUI.

`cmapLevs = [-2,-0.5,0,0.5,2]
cmapCol = ['black','black','red','red','black','black']
velGrndCmap, normVelGrnd = matplotlib.colors.from_levels_and_colors(cmapLevs,cmapVelCol,extend='both')`

gamaanderson, 2016-03-21:
thank you @stechma2 for the suggestion. The is an open issue (#126) to start using a new pyart functionality.

This is however one of those things that are super easy in text mode, but trick in GUI. We need to develop a concept on how to allow the user to select the colors. May be you can us with that.

stechma2, 2016-03-21:
Sorry for the duplicate issue - suppose I should have read back a ways further before posting.

Anyways, I will think about this and play with it as time allows and touch base with you all if anything comes to mind that hasn't already been suggested.

gamaanderson, 2016-03-21:
You don't need to be sorry. It is better to say something twice than to stay quiet.

nguy, 2016-03-21:
I asked @stechma2 to post a couple of issues we discussed over email so that we had some better tracking. Nice thing here is that we can easily tie the issues together through the github markdown interface.

nguy, 2016-05-05:
This was addressed in PR #142 . Is this okay to close?

stechma2, 2016-05-06:
I would say so. Haven't gotten a chance to play with the new version yet, but it certainly looks like it will do the trick!

nguy, 2016-05-06:
Okay, great. I'm closing the issue for now as the discrete colorbar has been added. If you find problems or see an improvement, maybe you could open a new Issue for that?

closed 2016-05-06 by nguy