nguy/artview #138: Aspect Ratio Option

stechma2, 2016-03-21:
If possible, it would be very useful to add an aspect ratio adjustment into the display options. In particular, I work with tail radar data primarily, and the default aspect ratio produces a very "squished" display which can be difficult to interpret.

gamaanderson, 2016-03-21:
ok, that can be an easy add. Can you post one image,  so that I have an better idea on how that is showing up to you?

gamaanderson, 2016-03-21:
but before that, have you tried "adjust display limits"? I know it is not optimal, but this should be able to unsquish your image.

stechma2, 2016-03-21:
Here is the image with the default aspect ratio, and display limits set to incorporate the entire observable domain:
[image]
Here is the same image, with the same display limits, but with aspect ratio set to 1 
(using this hard coded addition to `plot_radar`): `self.display.set_aspect_ratio(aspect_ratio=1,ax=self.ax)`
[image]
And finally, an image using the default aspect ratio, but adjusting the display limits and panning to unsquish and center the data. This method works, but is not ideal.
[image]

gamaanderson, 2016-03-21:
Ok, I see the difference, I will implement right away.

gamaanderson, 2016-03-21:
I just submitted a PR for that. The aspect radio can be found in the adjust limits.

thank you again @stechma2 for helping improving ARTView.

stechma2, 2016-03-21:
Great! Thank you for doing this @gamaanderson.

nguy, 2016-03-21:
Tangentially, I finally figured out why the file does not come up with the default Airborne x/y ranges. The *scan_type* value of the radar variable is not recognizing these particular files as airborne. I have a workaround that I'll comment on in the PR. We'll have this out in the very near future with this fix.

nguy, 2016-03-23:
Fixed with merge of PR #140

closed 2016-03-23 by nguy