nguy/artview #145: Vertical cross-section

nguy, 2016-05-17:
Create a vertical cross-section by adding a line to plot. Can use the existing index retrieval method in select region and then have the cross-section plot pop up in additional tab.

gamaanderson, 2016-05-18:
I agree with with the functionality, but I don't think we should use the the existing method, but rather just drawing the line in the axis and making the conversion from position in index to plot in a component for it self. This is mainly for performance and more flexibility. I already worked with cross section in an other software.

Should such a function of plotting a vertical section be make part of pyart? @jjhelmus?

jjhelmus, 2016-05-19:
I'd be open to including a method for extracting a vertical section from a radar volume to Py-ART.  Using a KD-Tree for gate lookup would be recommended.

swnesbitt, 2016-05-19:
Essentially, wouldn't this just require refactoring the pyart grid methodology, for cross sections with arbitrary cardinal direction orientations?  I suppose that great circle calculations would technically be appropriate, especially for longer cross sections.

jjhelmus, 2016-05-19:
@swnesbitt Yes, although I think it might be desired to do nearest neighbor interpolation instead of the weighted interpolation that the grid method currently uses.  Having a generic function in Py-ART which takes gates from one or more radars and a list of point and returns the fields values for those points using a weighted/nearest/k-nearest interpolator would be really useful in a number of cases.  The current gridding function does some of that but some work would be needed to expand the functionality.

nguy, 2016-05-20:
@gamaanderson @jjhelmus I've been working on a KD-Tree for another application and I agree that this probably would be a good implementation, especially for a polar coordinate data space. I'm finding that the KD-Tree can take a long time to build, just as a heads up (or I'm just messing up the programming).

Let's discuss implementation a little more. Building this in Py-ART and then calling in ARTview seems the idea. I agree with that plan. Where should this go @jjhelmus? If I can get my code working nicely, I can help get this up and running.

jjhelmus, 2016-05-20:
I'd put it in the `pyart/map` directory for the time being.  Once the details get fleshed out it may get moved but that seems reasonable for now.

gamaanderson, 2016-05-20:
So here we are again: good vs fast. A reformulation of the current `map_to_grid` or `map_gates_to_grid` should do a nice work. This would be fast enought if you follow the principe of drawing a line and then pressing a button for drawing. But not fast enought if you want to make the drawing in real time (while drawing), this is not only something that would look nice, but also the fast response create a 3 dimensional feelling.