nguy/artview #146: Keep user entered Nyqvist velocity

bdolan44, 2016-05-19:
We have found that when unfolding velocities, if you enter a Nyqvist velocity to unfold around, it is immediately reset to 0 as soon as you push a button to unfold, set up a boundary, etc. It would be great if the user entered value could remain. (This may only be a problem if the Nyqvist velocity is not in the radar file.)

nguy, 2016-05-20:
Thanks @bdolan44 we'll look into this. @gamaanderson can you take a look or do you want me to? I would guess it would be just retrieving the entered value and caching it in a class variable?

gamaanderson, 2016-05-20:
yes, the trick is to know when to use the variable and when to reset

nguy, 2016-05-20:
When the button is clicked to perform the operation, then we could grab the values entered and update?

gamaanderson, 2016-05-20:
and file change / field change?

nguy, 2016-05-20:
yep, I think we did this in something recently. Trying to remember which tool...

nguy, 2016-05-24:
Addressed in PR #149

closed 2016-05-24 by nguy