nguy/artview #147: Forward / back arrows lose all unsaved changed

bdolan44, 2016-05-19:
The other feedback I got about artview was that if you are unfolding a volume and you accidentally click the forward / back arrows (instead of up/down to go up in elevation), you lose any unsaved unfolding or editing that you've done. Kinda reminiscent of when solo would randomly crash with bus errors...but anyway. I know this is a challenge, but maybe even something like a prompt if you are in the 'modes' when you hit right/left arrow that asks 'Are you sure? Yes / No'

nguy, 2016-05-20:
I knew this was going to an issue that came up. There are a couple of different ways to handle this potentially. We need to decide what kind of direction would be best moving forward.

@gamaanderson the way I see it we have the following options. Feel free to suggest other options.

1. Have the tool save changes and not be reset by a change in a shared variable. This might suggest making some of the plugins their own shared variables.
2. Use the RadarCollection to store information about changes to each file in addition to stashing radar instances. I worry about this becoming less economical. My original thought was store many radar files and possibly images to speed loading files when switching back and forth.

gamaanderson, 2016-05-20:
your first point make sence if you want to preserve something like a select region. I'm not sure how to handle this, but the comment was about edits. In this case I would just add to the radar a modified=True attribute and when changing files popup if there are unsaved edits.

The use of a RadarCollection would not have this problem, since the files are not being closed. But this came at large cost.

nguy, 2016-05-20:
I like that idea, a check to save for the user. I do think that we could have "recall" on the plugins to use the the last state of usage. It would go away with restarting the program, but would stick around if the user went to the next file and then wanted to say select the same region as previous file.

I agree about the RadarCollection. This has been the sticking point with that idea, the cost of keeping many files in memory could kill the usefulness of the program.

nguy, 2016-05-24:
Addressed in PR #149

closed 2016-05-24 by nguy