nguy/artview #148: Manual filter - filter gates in radar stops working after a while

bdolan44, 2016-05-20:
After editing and unfolding a number of volumes, the "Manual Filter -Filter Gates in radar" button stops working. The fix is to exit out of Artview and restart.

gamaanderson, 2016-05-21:
is there any error mensage in the terminal?. Have you tried reseting the select region?

bdolan44, 2016-05-24:
We will get back to you on this....trying to reproduce the problem so we can give more specific details. However, resetting the select region does not fix the problem.

nguy, 2016-05-24:
Can you also not download the latest updated until you get more info and try the same steps after updating your code base? This might help rule things out, or not.
Thanks!

bdolan44, 2016-05-31:
Ok. I think we have not updated the latest. When this error occurs, this is what it says in the terminal:
File "/usr/local64/python-2.7/anaconda/lib/python2.7/site-packages/artview/components/tools.py", line 300, in onZoom
   relx = (cur_xlim[1] - xdata)/(cur_xlim[1] - cur_xlim[0])
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'

gamaanderson, 2016-05-31:
So, I would guess that xdata is None. I'm not really sure what caused that, but I would test for it and quit the `onZoom` function

gamaanderson, 2016-06-02:
So that error is created when the one try to zoom outside the picture, we should fix that, but I don't thing this is the bug here. So apparent it is not showing any error, that make things difficult.
One thing I have notice while trying to replicate is that if the selected region has to much components it gets slower and the program stop responding until it is done creating the Points. But after that it is back to normal.
I have created a small video with more information on the working of the select region: https://youtu.be/1Na5fY64m7Y

nguy, 2016-06-02:
I've noticed the same behavior in the past, but I'm not sure how to fix that. Nice video tutorial.

I wonder if we could simply raise an Exception on a None return?

nguy, 2016-06-02:
@bdolan44 Could you pull the latest changes and ensure they work for you as well. 
We'll have to look at the select region code again to see if it can be optimized. If you can recreate the problem, could you select *Reset Tools* from the Toolbox menu on the Display to see if this clears the problem?

bdolan44, 2016-06-07:
She was able to recreate the problem again and Reset Tools did not seem to fix the problem.

gamaanderson, 2016-06-07:
I could not recreate it, without more information there is not much I can do. I would need as much information as possible, really everything.

nguy, 2016-06-08:
Dang! We'll keep looking.

bdolan44, 2016-08-02:
We are still encountering this (occasionally). My student states: "I tried restarting Artview, clicking reset tools under toolbox, resetting my polygon, etc. etc., still won't filter out bad data. I'm not outside of the parts where I should be able to do this, either.  Here's the error I'm getting:

Traceback (most recent call last):
 File "/usr/local64/python-2.7/anaconda/lib/python2.7/site-packages/artview/plugins/manual_filter.py", line 181, in removeFromRadar
   data.mask[mask_ray, mask_range] = True
TypeError: 'numpy.bool_' object does not support item assignment

I don't know if that's any more helpful....

gamaanderson, 2016-08-02:
this error looks a little more helpful than before, at least it is in the part of the code it should be. I will give it one more look.

gamaanderson, 2016-08-04:
So, this appear to be a problem of the some what confusing / not well documented way numpy handle masked arrays. PR #160 may improve, but I'm not 100% sure.

One extra thing that I could do: this error is in a loop of fields, so we could add a `try:` there and if the error persist the function would still work in some fields. That however would create a silent error, so I'm not sure if it should be done.

nguy, 2016-08-05:
@bdolan44 I just accepted this fix from @gamaanderson 
Can you download the latest code base and test also?

bdolan44, 2016-08-08:
We will try it out and report back.

nguy, 2017-08-28:
@bdolan44 Did this get resolved?