nguy/artview #153: Egde option

gamaanderson, 2016-07-06:
add grafical interface of the `edge` option of the radar plot methods.

nguy, 2016-07-06:
I am working on this one now. I just need a file to test and then I'll create a PR for inspection.

nguy, 2016-07-06:
Will you take a look at PR#154?

I didn't add the code to the plot_grid. I think it should work the same, though. Correct?

gamaanderson, 2016-07-06:
yes, it could work just the same in plot_grid, this is however not much of a problem in the grid as it don't have the closing of azimuth problem.
I actually don't fully understand what it is the underling pyart error that create that problem, but if `edge=False` solves the problem, I'm fine with it.

nguy, 2016-07-06:
I believe it has to do with the interpolation that pyart performs to extract the edge from the gate center information. If there is wonky data then the interpolation gets a weird value. Is this correct @jjhelmus ?

gamaanderson, 2016-07-06:
for reference the original source of this issue: https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/artview-users/7gSpxa6_POM/MkRQGPwaBAAJ

jjhelmus, 2016-07-07:
> I believe it has to do with the interpolation that pyart performs to extract the edge from the gate center information. If there is wonky data then the interpolation gets a weird value.

The range, azimuth and elevation data stored in Py-ART's Radar class refers to the center of the range bin. If this is always set correctly is another question but typically even if the gates locations are specified differently it makes little perceivable difference.  Therefore for each gate we need to calculate the locations of the "edges" of  quadrilateral which will be have a center at the gate location.

For the range dimension this is relatively straight forward. For gates not at the endpoints, the edge is halfway between the gates. At the ends we extrapolate by adding or adding or subtracting half the distance from the previous/next gate.  

The azimuth/elevation dimension are a bit more difficult, as we need to take into account the possible periodic nature of the dimension and some physical limitation (elevation should be between 0 and 180 degrees) but the idea is similar, find the mid-point between rays and extrapolate at the edges.  This work good if the sampling of the azimuth or elevation is fairly uniform but if you have a big break or a wonky ray it can mess things up.  I'm guessing that is what is going in the linked example.

nguy, 2016-07-07:
Thanks for the explanation @jjhelmus.

nguy, 2016-07-07:
Issue closed in PR #154

closed 2016-07-07 by nguy