nguy/artview #155: Load time

gamaanderson, 2016-07-07:
as any graphical program artview takes some time to load, but it is kind to annoying me. We should check if there is no unnecessary loading or if there are things we can load just when we need it.

We could add some profiling to generate statistics of which functions are taking to much time. As we work on python it is important to optimise it once in while.

nguy, 2016-07-07:
I agree completely. We indirectly employ Py-ART lazy dictionary loading, but maybe we could do this more extensively with the internal code. 

We should also expand the testing....

gamaanderson, 2016-07-23:
in my system with empty cache it takes 17s to load pyart and just around 1.5s to the rest of artview. So we can not do much.

from those 17s, around 7s are just to load basemap. What about a lazy loading there @jjhelmus?

jjhelmus, 2016-07-23:
@gamaanderson Can you be more specific in what you mean by "loading".  Are you looking at the time it takes to `import pyart` or to generate a figure.  The former should not take 17 seconds, that later might if you are using basemap which can be quite slow.

gamaanderson, 2016-07-23:
I mean import, it does take so much, but for that I must empty the RAM cache. subsequent runs are faster.

nguy, 2017-03-24:
I did some profiling the other night. The pyart import remains somewhat slow, but I also noted that PyQt was loading very slowly it seemed. I don't have extensive evidence yet, but this is something to keep in mind and test in the future.