nguy/artview #163: New Layout

gamaanderson, 2016-09-06:
Unfortunate despite the holidays I have not been able to spend much time with artview. Nevertheless, I have worked forward in the new layout. I have changed several components to make them more compact, so they fit better in the dynamic layout. Those are present in my branch [layout](https://github.com/gamaanderson/artview/tree/layout) when using the layout script

    python artview -s layout

I have still not introduced the changes to artview core nor changed the standard script, but those would be the next step. The old version of some plugins are still available but shall also be removed. However I would like to first submit the changes to scrutiny, not sure how this shall happen.

There is one thing I didn't do, that is change the plugin `GateFilter`. I do not understand it and I would have done it different in the first place. So I don't think I have the authority to change it nor have I found a way to make it more compact preserving the functionality.

nguy, 2016-09-15:
Sorry, I'm now surfacing for air. I'd be happy to take a look at this. I wonder if we should take the layout slow and make this the v1.4 release? I think that v1.3 will have plenty to warrant the release in the somewhat near future. 

I'm working on the add text, but it was a little more difficult than I imagined. I hope to have that PR submitted in the next week or so time frame for your scrutiny.

gamaanderson, 2016-09-15:
ok, if you want to upload what you have no the text stuff I may take a look. Where are you having problems?

nguy, 2016-09-15:
It was with the structure more than anything and then I realized that it needed access to the display. I think I'm getting a solution in place, but also trying to give the option of retaining some information if desired.

nguy, 2016-10-09:
I finally got this working (i.e. had the time to get something finished). Check out PR #165 and leave some comments if you would.

gamaanderson, 2016-11-22:
I have continue working on this, I think it is pretty good. I have however abandoned the ideia of a soft introduction. In a small community there is no one trying to explore/ get used with optional configurations. People will just work with it where it become standard, so should just do it when we think it is ready.

nguy, 2016-11-23:
Yeah, let's take a look at it. Worst case scenario is we can roll back. But I bet it will be a vast improvement. Everything on my end is merged for now. I pulled that last bit in tonight.

gamaanderson, 2017-01-16:
just pushed a new commit in the branch, separating the open and the save menu. I thought the symbol could lead to confusion.

I would consider this to the be next PR, I do believe the new layout to be an improvement and I will allow us the design the appearance and the menus bars without losing coding independence. With this we will finally overcome the limitations I have (on purpose) added in the previews developments stages.

If you think this deserve more discussion before a PR, there are also the issue #164 and #173, where I already have some code waiting, those could be faster to merge.

gamaanderson, 2017-01-16:
If there is something you believe to be wrong, make a print screen. Remember there may be differences between  different operating systems.

nguy, 2017-01-16:
If you have code already prepped for PR #164  and #173 I'd prefer to merge those first and then iterate on layout. But I'm open to trying to get the layout first. Your call.

gamaanderson, 2017-04-06:
Merged

closed 2017-04-06 by gamaanderson