nguy/artview #164: Field calculation

gamaanderson, 2016-10-01:
I had an idea of how to use python magic to allow calculation and display of fields in a easy and intuitive way. something like:

    reflectivity *= 2

 Just wanted to register it here so I don't forget.

nguy, 2016-10-04:
Interesting idea!

gamaanderson, 2016-10-22:
I have it here [https://github.com/gamaanderson/artview/tree/radar_terminal]

I made a small (sound less video) to show the idea
https://www.youtube.com/watch?v=jXVg3g8cnck

nguy, 2016-11-15:
That is pretty neat and seems like it could be relatively easy to implement?

gamaanderson, 2016-11-15:
it use pretty depp [python magic](https://docs.python.org/2/reference/datamodel.html#special-method-names), but it is easy. The only thing I don't like is that we still depend on the system terminal, emulating one is more complicated.

nguy, 2016-11-16:
I agree it's not perfect, but better than nothing for the time being. I guess the argument could be made either way: If the tools are in place, porting may be simpler OR if we get too many tools in place, then porting is going to be a pain. Hmmm.

gamaanderson, 2017-01-29:
this was closed by PR #175

closed 2017-01-29 by gamaanderson