nguy/artview #168: PyQt5

pfhein, 2016-11-07:
The latest version of Anaconda Python 2.7 installs PyQt5 not PyQt4.  I just ran across this helping a student install artview on their Mac laptop.  It installs but then it can not find PyQt4.  Changing the 4 to 5 unfortunately did not work.  Here is the error when importing PyQt5 instead of PyQt4.
[image]

gamaanderson, 2016-11-08:
I have recently faced the same problem. 
My [master branch](https://github.com/gamaanderson/artview) is changed to support PyQt5, I'm not totally sure however how stable it is, would recommend nevertheless.

Merging this in the main project is problematic because of the extensive module re-structuration from PyQt4 to PyQt5. While we are working on this, I would suggest opening a branch PyQt5 in @nguy repository and add a notice in the readme.

pfhein, 2016-11-08:
Thank you @gamaanderson!  This was a big surprise to me.  I am glad that you are on top of it.

pfhein, 2016-11-09:
@gamaanderson - Unfortunately there are still issues.  I have not had chance to take a close look at it but here is error messages.
[image]

gamaanderson, 2016-11-09:
It does not seems you have the correct files.
This should be the (line)[https://github.com/gamaanderson/artview/blob/master/artview/core/core.py#L208]. Please try to make a plane new install

pfhein, 2016-11-09:
I had grabbed the zip file, so maybe that was a problem, and I did not clean out the old install.  I will try again.

nguy, 2016-11-15:
I recently ran into this when I changed jobs and was setting my computer up.

Another work around is to downgrade your PyQt to PyQt4.
You can get a list of packages: `conda search pyqt`
And then use the info to install a version 4 distribution: `conda install pyqt=4.11.4`

I think this should work, but as @gamaanderson mentioned, we'll have to make the move eventually.

nguy, 2016-11-22:
@pfhein @gamaanderson Where are we at on this issue?

gamaanderson, 2016-11-22:
If I got the time, tomorow I will make a PR. This will be a high conflict PR, so if you have anything not publish you should present it, so I can solve any conflicts.

nguy, 2016-11-22:
Okay, there is a PR implementing the small fix you pulled back and a lot of small pep8 maintenance. If you could just breeze through quick and verify, that would be great. I can then hold off any more changes until you address this issue. In the meantime, I'll try to start commenting on some posts to get an idea of what we should work on next.

gamaanderson, 2016-11-23:
implemented in PR #171

nguy, 2017-01-17:
Can this topic be closed? 
I've opened another issue #174 to address the lingering problem with Qt5 compatibility.

pfhein, 2017-01-17:
Yes, it can be closed.  I will do it.

closed 2017-01-17 by pfhein

nguy, 2017-01-18:
@pfhein I'd love to hear any issues you run across. Make sure PyQt4 is available on machines.