nguy/artview #172: Bug fixes

nguy, 2016-12-06:
In testing the last few updates I've identified a few things that I think need to be fixed:
[ ] Change the ManualFilter name to ManualEdit. I believe this is the nomenclature most people know.
[ ] When the region select is turned off, the lines continue to display on plot. When any Mode or Plugin that uses the region selection tool is closed. Remove lines from the plot and redraw the plot.
[ ] Update the _check_entries method in image_text to be more robust.

gamaanderson, 2016-12-06:
the first is easy, the third I don't know what else you want there. 
The second one is not that clear, since region select is a plugin it self. Nevertheless to make clean up one can re-implement the [close event](http://pyqt.sourceforge.net/Docs/PyQt4/qwidget.html#closeEvent)

nguy, 2016-12-07:
What you did will certainly work.
I think we may want to re-institute, but I'm not sure quite how yet.

gamaanderson, 2017-01-29:
should we add addressing this as a requirement to version 1.3?

nguy, 2017-02-03:
Definitely the first two items, they should be easy fixes. The 3rd item I think we should put in the same list as the last two PRs that gave structure and will get refinement. We could start a separate issue for post v1.3?

gamaanderson, 2017-03-10:
commit https://github.com/gamaanderson/artview/commit/ca01c80d24378f41e43100df9d7fc4385552ccda address this issue

closed 2017-03-24 by nguy