nguy/artview #178: Unfolding stops at 0

pfhein, 2017-01-26:
I have seen a case where the unfolding/deliasing stops at zero velocity.  The velocity is large enough that is wraps around and goes past the 0m/s mark. but the unfolding does not catch it.

gamaanderson, 2017-01-27:
isn't this a pyart issue?

for reference the desiasing is done either with:
    pyart.correct.dealias_region_based(radar, gatefilter)
    pyart.correct.dealias_unwrap_phase(radar, gatefilter)
depending on which plugin you are using

if you are talking about the manual unfold, such a case can not be corrected with that simple of an algorithm, I recommend using the pyart methods before.

pfhein, 2017-01-27:
No, This is manual unfolding/deliasing so not the automatic dealiasing of pyart.  Thinking about this, I wonder that it might not be possible to cross the 0 line since that would make it harder to distinguish between those points that do not need to be dealiased and those points that do.  (If you are really careful selecting the boundary you can do it.)  

Here are two plots.  One before the dealiasing and one after.  The dealiasing happens in the upper right quadrant of the plot.
Before dealiasing
[image]
And the after dealiasing
[image]
You can see that the discontinuity remains but it was shifted to the 0m/s contour.

gamaanderson, 2017-01-27:
I'm not really sure how we can solve this, since as you said the zero line is what is being used to separate what need dealing and what does not. We could allow the user to change this threshold value, but I'm not sure about that. We could also add a component to do a linear transformation in a selected region + gatefilter, that would probably be much more difficult to work with, but gives you full control.

Let me know what you think and we may add as a to do, and it will hopefully be addressed at some time. Also if you want to do it your self I would be glad to give directions.

gamaanderson, 2017-01-27:
by the way @pfhein I'm planning to be presenting ARTView in the AMS's 38th Conference, on august. If you have any thoughts I would be glad to hear, send me an e-mail.