nguy/artview #181: Combine Display and Modes SelectRegions

nguy, 2017-02-17:
I think we should try to combine these two feature into a single code base or at least call the same underlying code so that they do not diverge. Also I think the following should be addressed:

1. When a new polygon is selected, open tables/graphs should automatically update
2. Add the stats/tables/plot option to the Manual filter. I think this would be useful when editing data in this way.
3. Merging these together, I think should take care of the fact that if you create a polygon with the Display SelectRegion, it does get removed when the Modes SelectRegion is used.

gamaanderson, 2017-03-12:
try to address that in branch https://github.com/gamaanderson/artview/tree/issue181

nguy, 2017-03-24:
I believe this can be closed also?

gamaanderson, 2017-04-05:
closed by PR #186