nguy/artview #184: Metadata not updating in plot

swnesbitt, 2017-03-14:
I'm perusing some NPOL RHI volumes, and the titles of the plot are not updating with the correct information when I change sweeps.  Note inconsistency between drop down and titles.

Data plot itself seems to update when I select a new sweep.

[image]

gamaanderson, 2017-03-14:
@swnesbitt thank you for reporting this.
This should not happen, so I will have to take a better look. If you know how to use pyart, could you give use what [pyart.graph.common.generate_title](http://arm-doe.github.io/pyart-docs-travis/dev_reference/generated/pyart.graph.common.generate_title.html#pyart.graph.common.generate_title)  returns. We take the title from there.
In the mean while you can change the title in the display options.

gamaanderson, 2017-03-16:
I could not replicate this. What version are you using?

nguy, 2017-03-23:
@swnesbitt Can you check two things:
1. Is the title updating when you select different tilts( just not updating the text)?
2. This looks like RHI data. Are the tilts actually azimuths that the data is along? Could the reader underneath be doing something funky translating the data?

nguy, 2017-06-29:
Wondering if this was still an issue?

swnesbitt, 2017-06-29:
Seems to be working when I pulled the repository this morning.

-Steve

-- 
Steve Nesbitt, Associate Professor
Department of Atmospheric Sciences, University of Illinois at
Urbana-Champaign
Room 3060 Natural History Building, 1301 West Green Street, MC-104, Urbana,
IL 61801 USA
voice : +1.217.244.3740  fax : +1.217.244.1752
internet : snesbitt@illinois.edu ;
https://www.atmos.illinois.edu/people/snesbitt ;
http://publish.illinois.edu/snesbitt

closed 2017-08-28 by nguy