nguy/artview #193: Background Map not working now

nguy, 2017-04-11:
I recently tested the map background module. I get the following error:

```
Traceback (most recent call last):
  File "/Users/guy/anaconda/lib/python2.7/site-packages/artview/core/core.py", line 284, in closeEvent
    self.disconnectAllVariables()
  File "/Users/guy/anaconda/lib/python2.7/site-packages/artview/core/core.py", line 247, in disconnectAllVariables
    self.disconnectSharedVariable(var)
  File "/Users/guy/anaconda/lib/python2.7/site-packages/artview/core/core.py", line 266, in disconnectSharedVariable
    self.sharedVariables[var])
TypeError: 'instancemethod' object is not connected
Fallingback to pbuffer. FBO status is 36054

Fallingback to pbuffer. FBO status is 36054

Fallingback to pbuffer. FBO status is 36054

Traceback (most recent call last):
  File "/Users/guy/anaconda/lib/python2.7/site-packages/artview/plugins/background.py", line 90, in apply
    from mpl_toolkits.basemap import shiftgrid, cm
  File "/Users/guy/anaconda/lib/python2.7/site-packages/mpl_toolkits/basemap/__init__.py", line 30, in <module>
    from mpl_toolkits.basemap import pyproj
ImportError: cannot import name pyproj
```

Here's the package environment:
```
# packages in environment at /Users/guy/anaconda/envs/py35_qt4:
#
curl                      7.49.0                        1  
cycler                    0.10.0                   py35_0  
freetype                  2.5.5                         1  
hdf4                      4.2.12                        0  
hdf5                      1.8.17                        1  
icu                       54.1                          0  
jpeg                      8d                            2  
libnetcdf                 4.4.1                         0  
libpng                    1.6.27                        0  
matplotlib                1.5.1               np111py35_0  
mkl                       2017.0.1                      0  
netcdf4                   1.2.4               np111py35_1  
numpy                     1.11.3                   py35_0  
openssl                   1.0.2j                        0  
pip                       9.0.1                    py35_1  
pyparsing                 2.1.4                    py35_0  
pyqt                      4.11.4                   py35_4  
python                    3.5.2                         0  
python-dateutil           2.6.0                    py35_0  
pytz                      2016.10                  py35_0  
qt                        4.8.7                         4  
readline                  6.2                           2  
scipy                     0.18.1              np111py35_1  
setuptools                27.2.0                   py35_0  
sip                       4.18                     py35_0  
six                       1.10.0                   py35_0  
sqlite                    3.13.0                        0  
tk                        8.5.18                        0  
wheel                     0.29.0                   py35_0  
xz                        5.2.2                         1  
zlib                      1.2.8                         3  
```

It appears that pyproj has been deprecated. That said, this gives us the opportunity to switch to cartopy which I feel is now more robust and gaining a larger user base.
This should either be disabled for v1.3 or fixed. And switch to cartopy for v1.4

mustafaalicelik, 2021-03-23:
Does Background map working correctly now?

gamaanderson, 2021-03-23:
It does not seem that we changed anything, but it was also not an error in our code, but in mpl_toolkits. Anyway I just tested, and it worked in my machine (a debian with python3), but it is far from being user-friendly. Write if you need further help in using it, I don't think there is anywhere explaining it.