nguy/artview #197: run over database

gamaanderson, 2017-05-06:
we could create an component to run over database and search for radar from date and time.

This would work as an example of what can be done, as it is user dependent. @CesarBeneti @TulipaSilva any interest?

nguy, 2017-05-25:
Not really a database, but I have automated code that goes out to the NOAA AWS bucket and retrieves Level 2 data. I have hesitated putting this in due to adding more dependencies. But I will likely try prior to the radar conference as it should be straightforward. (Though I'd like to get v1.3 released first!)

It probably depends upon what type of database you want to query. You'll likely need some specific packages for different archetypes. 

This might lend itself to the RadarCollection development as well that we discussed previously.

gamaanderson, 2017-05-26:
I don't want it to be necessary a fix part of artview, I manly want an example to show off and can be adapted to the specific needs o users.