nguy/artview #208: Problems with pyQt4

fernandomhahne, 2017-10-18:
I have installed artview through Anaconda on Debian, but when I try to run it, I get the following error:

```
Traceback (most recent call last):
  File "/home/fmhahne/.anaconda2/bin/artview", line 4, in <module>
    import artview.__main__
  File "/home/fmhahne/.anaconda2/lib/python2.7/site-packages/artview/__init__.py", line 54, in <module>
    from . import core
  File "/home/fmhahne/.anaconda2/lib/python2.7/site-packages/artview/core/__init__.py", line 21, in <module>
    from . import common
  File "/home/fmhahne/.anaconda2/lib/python2.7/site-packages/artview/core/common.py", line 8, in <module>
    from .core import QtGui, QtCore
  File "/home/fmhahne/.anaconda2/lib/python2.7/site-packages/artview/core/core.py", line 10, in <module>
    from PyQt4 import QtGui, QtCore
ImportError: No module named PyQt4
```

I tried to install pyqt4 (pyqt5 is already installed) with:

```
conda install pyqt=4
```

but I get:

```
Fetching package metadata ...........
Solving package specifications: .

UnsatisfiableError: The following specifications were found to be in conflict:
  - arm_pyart
  - pyqt 4*
Use "conda info <package>" to see the dependencies for each package.
```

Someone knows how to fix it?

gamaanderson, 2017-10-18:
I would suggest using the newest version (1.3) which is PyQt5 compatible. There is no conda package for it, so you will have to download from here and install manually (it should be really simple).

fernandomhahne, 2017-10-18:
It worked. Thanks!

closed 2017-10-18 by fernandomhahne