nguy/artview #209: Dumb installation question

njmessina, 2019-02-12:
Hi there... do I need to have pyart already installed before installing artview using your conda channel?  I just installed Anaconda, then typed:
conda install -c jjhelmus artview
.... it's taking a very long time...
thank you...
joe

gamaanderson, 2019-02-13:
If I am not mistaken, conda will try to install the dependences, that mean
it will install PyArt and its dependences. That should in fact take some
time. How long is it taking?
That is however not my field of expertise, I hope Nick can give more
information on that.

Cheers
Anderson

njmessina, 2019-02-13:
Thanks…
I just tried installing arm_pyart on a brand new Anaconda installation.  It causes Spyder to quit working and I can’t figure out why.

Macs-MacBook-Air:bin joe$ spyder
/Users/joe/anaconda3/bin/pythonw: line 3: /Users/joe/anaconda3/python.app/Contents/MacOS/python: No such file or directory

It worked fine before I ran 'conda install -c conda-forge arm_pyart’.  Grrrrr.   Any advice would be great!

thanks again…
Joe

nguy, 2019-02-13:
Hi Joe,

First off for the long install. I think this is something going on with the conda channels. Conda is what is called a package manager and will search for and install dependencies needed for the artview install for example. I've noticed the last couple of days that this has been really slow for me, taking up to 10 minutes for some installs. And the combination of artview and pyart does end up having a good number of dependencies.

Per your spyder question. I'm not familar with spyder, but for a fresh conda environment, you might need to install spyder as well for that environment. If you installed in another env before (not root or base), then it won't be accessible from your new environment.

njmessina, 2019-02-13:
Hi Nick and Anderson,
I was able to fix the Spyder problem by doing a fresh install of Anaconda with Python 3.6, then installing arm_pyart.
However, when I went to install Artview (using the ‘conda’ install), it wiped out Python 3.6 and left me with Python 2.?.
I can use Artview on another machine here at work, so not a complete disaster.  Pyart and Spyder are more important for me, but I thought you guys should know what’s happening because I’m trying to put Artview and pyart on ‘clean’ installs of Anaconda.
I’ve only been using Python for a little over a year and still don’t really understand what environments are, so I only have one.  I’m using Mac OS X 10.11.6 if that makes any difference.
Thanks for your time!
Joe