nguy/artview #213: A working conda environment

aschueth, 2019-09-15:
Hi,

I'm trying to install artview, but I continue running into issues with the dependencies. Namely, when I install from conda 'conda install -c jjhelmus artview', I get an unsatisfiable error, even with a clean environment. I've tried python 2.7, 3.4, 3.6, and installing packages first as well as installing artview first and packages later. Everytime, it fails to install artview. 

I'm wondering is the jjhelmus channel artview stable, and if so is it possible to get a stable working enviornment .yml for it? Otherwise do I need to install from source or is artview not being maintained as much anymore?

dopplershift, 2019-09-16:
It looks like the latest is available on the conda-forge channel as well.

aschueth, 2019-09-16:
Yes indeed, probably should've checked there first. Would it be worth adding to the README that the correct version should be from conda-forge instead of the jjhelmus channel?

gamaanderson, 2019-09-17:
Thank you for solving this. I made a pull request to update the README

nguy, 2019-09-18:
Thanks for the report @aschueth and the suggestion @dopplershift 
@gamaanderson fixed this in a merge request that updated the README file

closed 2019-09-18 by nguy