nguy/artview #215: Conda-forge v:1.2 on linux, v:1.3.7 on mac

aschueth, 2019-10-03:
I have another issue when it comes to installing artview on different platforms. I finally have a working environment on my mac, however when transferring a .yml file to linux it failed rather impressively. Installing package by package I noticed that the only versions of artview on conda-forge available for linux is 1.2. With 1.2 I am having pyqt issues and so I was hoping to upgrade to the working 1.3.7 version. When I searched the available artview versions on mac (conda search artview --channel conda-forge), I got 1.2 and 1.3.7. I was wondering if there was a working version of artview 1.3.7 for linux and if that could be put on conda-forge?

dopplershift, 2019-10-07:
It looks like the recipe for conda-forge has gotten a little...stale. Probably best to close this one and open over at: https://github.com/conda-forge/artview-feedstock

aschueth, 2019-10-07:
Ahh, didn't know how all that worked. I'll open it up over there and close this issue.

closed 2019-10-07 by aschueth

nguy, 2019-10-08:
@dopplershift "Stale" is very kind. I'm working on scoping some new work I hope in the new year. I'd like to bring this back to life more fully.