nguy/artview #218: Add reading files auto

mustafaalicelik, 2021-01-11:
I want to add a func to standart script, it will read the files from my radar file directory one by one auto with a specific time. I'm open for any suggestions.

gamaanderson, 2021-01-11:
It depends on what you mean by read the files. If you mean open them and story in memory, not sure if possible. There were some talks of implementing something like that, but I don't remember it being done, mainly because it would fill you memory pretty quickly.
If you mean to have those as a list of filename, so that you can switch quickly, then what you want is to load a navigator, as the main menu just handle one radar at a time.
The script receives a variable `DirIn` from the `-d` command line option, you could use this to get a list of files. It would probably be something like this

    filelist = ____________ # use the sys to get a list of string with fullname of Radar files in DirIn
    Vfilelist = Variable(filelist)  # create shared variable
    navigator = Navigator(pathDir=dirIn, filename=None, Vradar=MainMenu.Vradar, Vgrid=MainMenu.Vgrid, Vfilelist=Vfilelist, name="FileNavigator", parent=MainMenu)  # start the navigator with the Vfilelist and connect to Mainmenu Vradar and Vgrid
    view.MainMenu.addLayoutWidget(navigator) # this should prevent the navigator being in a different window
    
I did not test this and it has be long since I handle with this things, so I may be overlooking something

gamaanderson, 2021-01-11:
Please do not be afraid to ask  questions, I will gladly help, I may just need some time to answer if I need to start writing and testing code.

mustafaalicelik, 2021-01-11:
I sent you a mail, ill be glad if you can help.