nguy/artview #219: AdvanceFileSelect not working

mustafaalicelik, 2021-01-13:
This lines from "artview.components.menu.Menu" :

`    def AdvanceFileSelect(self, findex):
        '''Captures a selection and open file.'''
        if findex > (len(self.Vfilelist.value)-1):
            msg = "End of directory, cannot advance!"
            common.ShowWarning(msg)
            findex = (len(self.Vfilelist.value) - 1)
            return
        if findex < 0:
            msg = "Beginning of directory, must move forward!"
            common.ShowWarning(msg)
            findex = 0
            return
        self.fileindex = findex
        self.filename = self.Vfilelist.value[findex]
        self._openfile()`

"Beginning of directory" is working but "End of directory" part is not working. What is wrong with these lines?

gamaanderson, 2021-01-13:
what do you mean by not working? What exactly is happening?

mustafaalicelik, 2021-01-14:
Nothing, it advance to the first file from dir. No error msg shown.

gamaanderson, 2021-01-14:
the "returns" are wrong, and they have being wrong all the way. But this case should never be executed, since the buttons for it should not been enabled to begin with. I will eventually fix it, now that I got artview running in my new laptop.

mustafaalicelik, 2021-01-14:
When I open a random file from dir and using left key to open prev file, the "Beginning of the directory, must move forward" msg occurs. When I use right key to open next file, its opening the first file in directory. I think its not indexing the directory correctly. It should understand the first and the last file right after opening file from a directory.

gamaanderson, 2021-01-14:
in this case I presume there is an error in the filelist. Probably the same error we just talked about.

gamaanderson, 2021-01-14:
@nguy do you have a windows PC running artview? Could you test if moving from files within a directory is working?

closed 2021-01-18 by mustafaalicelik